drewsum/Electronic_Display #65: Rotary Encoder Module

drewsum, 2019-02-02:
Should include init function, interrupts to detect rising and falling edge of ENCODER_STEP signal, and global variables to keep track of desired brightness.

cmgilger1, 2019-03-26:
currently have interrupt to detect rising edge for ENCODER_STEP signal in misc_board_control.c, should I move that ISR into rotary_encoder.c?

I haven't pushed to master yet, but this is what the rising edge ISR handler looks like... 
```
void encoderStepREHandler(void) {
   
    uint8_t current_brightness = (OC4R * 100) / PR2;
    
    // if ENCODER_DIR_PIN is high, we are rotating clockwise
    if (ENCODER_DIR_PIN) {
        
        // If we are not at maximum brightness, make brighter
        if (current_brightness < 100) {
            
            panelPWMSetBrightness(current_brightness + 10);
            
        }
        
    } else {
        
        // if we are not at minimum brightness, make dimmer
        if (current_brightness > 0) {
        
            panelPWMSetBrightness(current_brightness =- 10);
            
        }   
        
    }

}
```

I know it probably needs changes but I'm thinking we should put this in the rotary_encoder.c file

drewsum, 2019-04-02:
This works, awesome job. Closing.

closed 2019-04-02 by drewsum