drewsum/Electronic_Display #138: Solve Power Board Input Protection MOSFET Overheating

drewsum, 2019-04-12:
We've been using the IPC100N04S5-1R7 as the input protection MOSFET on the power board. Under load it's getting too hot. If this MOSFET fails, the entire system fails. We need to pick a new MOSFET with lower RDS(on) that will not overheat.

Needs to have VGS > +/- 20V, IDS > 50 Amps, RDS(on) < 1mOhm, VDS > 20V

cmgilger1, 2019-04-12:
https://www.digikey.com/product-detail/en/infineon-technologies/IPC100N04S51R2ATMA1/IPC100N04S51R2ATMA1CT-ND/6578406

cmgilger1, 2019-04-12:
https://www.digikey.com/product-detail/en/infineon-technologies/IPC100N04S5L1R1ATMA1/IPC100N04S5L1R1ATMA1TR-ND/6559884

VGS = +/- 16V
IDS = 100A
RDS(on) = 1.1mOhm (VGS=10V, IDS=50A)
VDS = 40V

drewsum, 2019-04-18:
Added two beefy heatsinks to the input protection FETs, donated by Dean and Nate. Will want to replace FETs with the lower Rds(on) alternatives we bought later anyway.

cmgilger1, 2019-04-22:
parts came in and I'll be putting them in the locker tomorrow morning

drewsum, 2019-04-27:
The original input protection FETs we have installed on the power board (IPC100N04S51R7ATMA1) have a voltage drop of 2 Volts with a white image at full brightness. This explains why they get so damn hot. Running the numbers, I'm guessing the two series FETS have a combined RDSon of 120 mOhm. That's waaaaay too high. Trying to swap some FETs to see the other results we can get.

drewsum, 2019-04-27:
I must have been smoking some crack when I picked the original input protection MOSFETs.  I replaced both of them with IPC100N04S51R2ATMA1 and it runs great, really cool. Only a 19mV voltage drop at full load. Let's stick with these. No heatsinks needed

closed 2019-04-27 by drewsum